Thinkersbluff/DGUS-Reloaded_for_CR6-DWIN-SET_Component #37: SAVE_CONFIG fails if the display brightness setting is changed

Thinkersbluff, 2023-08-27:
To reproduce this problem:
1. Restart or Firmware Restart Klipper
2. On touchscreen, press Setup.
3. On Setup screen, adjust brightness value.
4. Enter SAVE_CONFIG from console 
5. Klipper reports: "SAVE_CONFIG section  't5uid1' option 'brightness' conflicts with included value
Note: even after adjusting brightness back to the default value (100 per the DGUS_Reloaded.cfg file), SAVE_CONFIG fails with the same error.

**Work around**: 
1. Comment-out (or delete) this line in DGUS-Reloaded.cfg:  brightness: 100
2. Restart or Firmware Restart Klipper
The screen will now remain at the last selected brightness level, even if power-cycled or if RESTART or FIRMWARE RESTART is used.

Thinkersbluff, 2023-08-27:
This issue will be resolved in DGUS-Reloaded Klipper Component v1.2.9.

Thinkersbluff, 2023-08-27:
Corrected with commit a81298a on branch New-At-1.2.9 of the Klipper component.

Thinkersbluff, 2023-08-28:
A second review reveals that SAVE_CONFIG works fine, if the [t5uid1] section is moved back to the printer.cfg file, instead of including it by including DGUS-Reloaded.cfg.  
Looks like an earlier decision to create and include DGUS-Reloaded as a quick way of disabling/enabling DGUS-Reloaded has run afoul of the "Law of unintended consequences".  Moving [t5uid1] back to printer.cfg may be the best way to solve this.

txigreman, 2024-03-05:
I saw [your discussion here]( https://klipper.discourse.group/t/how-set-save-config-pending-items-to-in-a-macro/10203 ), and IMHO, I don't like how `SAVE_CONFIG` works.

`SAVE_CONFIG` just looks for stuff in `printer.cfg` file, comments out overrode config, and appends an ugly block of commented code, with a magic header, at the end of the file. If it can't find the property it's trying to override in that file (excluding includes), it fails miserably.

I think a much better approach would be to have some kind of `printer.override.cfg` file where `SAVE_CONFIG` could put all this overrode properties, and don't mess with my `printer.cfg` file. The process would be as simply as:

1. Load config from `printer.cfg` and includes.
2. Override it with values present in `printer.override.cfg`.

Anyhow, we are limited by the technology of our time, so it seems that the only option to make it work is, as you said, to put that block of config in the `printer.cfg` file.

txigreman, 2024-03-05:
> I think a much better approach would be to have some kind of `printer.override.cfg` file where `SAVE_CONFIG` could put all this overrode properties, and don't mess with my `printer.cfg` file. 

I've raised this as a (feature request in the klipper forum]( https://klipper.discourse.group/t/save-config-in-separated-file/14679 ), but I have 0 expectations (or lower).

closed 2024-10-19 by Thinkersbluff