Thinkersbluff/DGUS-Reloaded_for_CR6-DWIN-SET_Component #52: Custom macros

jensweimann, 2025-01-12:
Hi, 

thank you for this great project! I have some custom macros and it would be great to have a custom page where I could call them from the screen. Would it be possible?

Thinkersbluff, 2025-05-01:
That's an interesting idea.  Beyond my current skillset to program, unfortunately.  

Not helpful, I suppose, but you can run all macros from the Mainsail console...
To help me and others think about it, can you provide some examples of how/where these macros would fit into the workflow?

jensweimann, 2025-05-01:
Well right now, I have these custom macros for myself:

<pre>
<details>
  <summary>my_macros.cfg</summary>
 ``` 
[gcode_macro START_PRINT]
description: Set printer config parameters. Load BED_MESH profile and Z Offset based on nozzle target temperature.
gcode:
  {% set toolTemp = params.EXTRUDER_TEMP|int %}
	{% set bedTemp = params.BED_TEMP|int %}
	{% set toolProbeTemp = 145 %}
	
  M300 # BEEP at start
  REPORT_UNLOAD_FILAMENT_AFTER_PRINT
  DGUS_PRINT_START  # Causes DGUS-Reloaded to switch to the Print Status screen
  SET_DISABLE_PAUSE_STOP # Tell DGUS-Feloaded to disable the PAUSE and STOP buttons until the purge line is being drawn. (Solves Issue#31)
  STORE_REPRINT_PARAMETERS  # Saves the current print job name and path, to enable reprinting this job by running the macro REPEAT_LAST_PRINT

  BED_MESH_CLEAR
  BED_MESH_PROFILE LOAD=default # load default mesh profile at startup
  
  # start heating as soon as possible
  SET_HEATER_TEMPERATURE HEATER=extruder TARGET={toolProbeTemp}
  {% if bedTemp >0 %}
  SET_HEATER_TEMPERATURE HEATER=heater_bed TARGET={bedTemp}
  {% endif %}

  M82 								# absolute extrusion mode
	M220 S100 							# Reset Feedrate
	M221 S100 							# Reset Flowrate

  # If meant to print without heating bed, need to prevent software waiting for bed to cool to zero!
  {% if bedTemp >0 %}
    TEMPERATURE_WAIT SENSOR=heater_bed MINIMUM={bedTemp-10} MAXIMUM={bedTemp+10}
  {% endif %}
  TEMPERATURE_WAIT SENSOR=extruder MINIMUM={toolProbeTemp-5} MAXIMUM={toolProbeTemp+5}
  
  # homing and probing
  # this is needed to get rid of electrical side effects on the probe
  SET_HEATER_TEMPERATURE HEATER=extruder TARGET=0
  G28
  SET_HEATER_TEMPERATURE HEATER=extruder TARGET={toolTemp}

  # Set movement to Absolute Positioning mode
  G90

  # Return to left front corner after homing Z, with Z height at 20mm 
  G0 Z20
  G0 X0 Y0 F5000
  SET_HEATER_TEMPERATURE HEATER=extruder TARGET={toolTemp}
  TEMPERATURE_WAIT SENSOR=extruder MINIMUM={toolTemp-2} MAXIMUM={toolTemp+5}
    
  # Re-enable the Print Pause and Stop buttons on the Printing screen
  CLEAR_DISABLE_PAUSE_STOP

	# And we can now draw our purge line
	G92 E0 								#Reset Extruder
	G1 Z2.0 F3000 						#Move Z Axis up
	G1 X0 Y20 Z0.28						#Move to start position
	G1 X0 Y200.0 Z0.28 E15   			#Draw the first line
	G1 X0 Y200.0 Z0.28 					#Move to side a little
	G1 X0.5 Y20 Z0.28 E30 				#Draw the second line
	G92 E0 								#Reset Extruder
	G1 Z2.0 F3000 						#Move Z Axis up

  # IF the RunoutSensor is Enabled, and it is reporting No Filament Detected, then Pause and prompt for filament, ELSE start printing
  TEST_RUNOUT_STATE

[gcode_macro END_PRINT]
[... 116 lines cut ...]

[gcode_macro REPORT_UNLOAD_FILAMENT_AFTER_PRINT]
gcode:
  {% set svv = printer.save_variables.variables %}
  {% if svv.unload_filament_after_print == 1 %}
    M117 unload_filament_after_print enabled
    M118 unload_filament_after_print enabled
    SET_PIN PIN=rpi_led value=1
  {% else %}
    M117 unload_filament_after_print disabled
    M118 unload_filament_after_print disabled
    SET_PIN PIN=rpi_led value=0
  {% endif %}

[delayed_gcode STARTUP_REPORT_UNLOAD_FILAMENT_AFTER_PRINT]
initial_duration: 1.0
gcode:
  REPORT_UNLOAD_FILAMENT_AFTER_PRINT
 ``` 
</details>
</pre>

In short, I would like to enable auto-unload-filament-after-print from the touchscreen and my custom load/unload-filament macros.

I also have some lights around the printer that I could configure as macros and trigger them from the touchscreen.

Thinkersbluff, 2025-05-01:
I will think on this for a bit.

For the Start Print, End Print and Run ABL macros, you can just edit the default macros in CR6.cfg, right?.

For the Enable Auto Unload Filament after Print macro:

- Would you want to run the "enable" macro during a print? (e.g. from Print Tune menu)
- Would you also want to be able to set that as your default behaviour (e.g. on SetUp menu) 
-  or on a print by print basis (i.e. on Prepare Menu)?

Would it be ok to have an on/off button like the Runout Sensor Enable/Disable button, for this function?

Thinkersbluff, 2025-05-01:
The Load/Unload filament macros are a tad trickier to integrate into the current architecture, just because Desuuu coded them directly into the vars-in.cfg python script.

I see some irrelevant checks for multiple extruders, which you could drop.
He reads parameters like nozzle temperature and filament length from the screen variables, tests whether the button pressed was for Load or Unload, then runs the applicable G-code.  You could probably replace that G1 instruction with the name of your custom macro, to integrate that.

[t5uid1_var __filament_move]
type: input
address: 0x201e
data_type: uint16
script:
  {% if is_busy() %}
    {% do set_message("Busy") %}
  {% else %}
    {% set t5uid1 = printer.t5uid1 %}
    {% set extruder = get_variable("filament_extruder",
                                  t5uid1.constants.extruder_current) %}
    {% if extruder == t5uid1.constants.extruder_current %}
      {% set ext_name = printer.toolhead.extruder %}
    {% elif extruder == t5uid1.constants.extruder_e0 %}
      {% set ext_name = "extruder" %}
    {% elif extruder == t5uid1.constants.extruder_e1 %}
      {% set ext_name = "extruder1" %}
    {% endif %}
    {% if ext_name not in printer %}
      {% do set_message("Invalid extruder") %}
    {% elif printer[ext_name].temperature < heater_min_extrude_temp(ext_name) %}
      {% do set_message("Temperature too low") %}
    {% else %}
      SAVE_GCODE_STATE NAME=state_filament_move
      ACTIVATE_EXTRUDER EXTRUDER={ext_name}
      M83
      {% set length = get_variable("filament_length", 150) %}
      {% if data == t5uid1.constants.filament_retract %}
        G1 E-{length} F1800
      {% elif data == t5uid1.constants.filament_extrude %}
        G1 E{length} F400
      {% endif %}
      RESTORE_GCODE_STATE NAME=state_filament_move
    {% endif %}
  {% endif %}
run_as_gcode: true

Thinkersbluff, 2025-05-01:
It took me a while to figure out how to build the print menu from the Virtual SD card.  
I suspect that implementation is still a bit glitchy, depending on what you do to that card directly from Mainsail or Kli[pperScreen.

I have wondered before about whether I could implement a similar function as a Macros page, to scroll through available macros and  run any that did not require further interaction.  

That would seem to cover the more general case for custom macros support, but would be "dangerous" to access from the Print Tune menu.

jensweimann, 2025-05-02:
> * Would you want to run the "enable" macro during a print? (e.g. from Print Tune menu)
Yes, that would be great.

> * Would you also want to be able to set that as your default behaviour (e.g. on SetUp menu)
I guess that would be kind of surprising if you start a print after a couple of weeks and forget about that feature?

> * or on a print by print basis (i.e. on Prepare Menu)?
For me at least that is perfectly sufficient.

> Would it be ok to have an on/off button like the Runout Sensor Enable/Disable button, for this function?
Yes, sure, that sounds good.

jensweimann, 2025-05-02:
> It took me a while to figure out how to build the print menu from the Virtual SD card. I suspect that implementation is still a bit glitchy, depending on what you do to that card directly from [...]

The

> It took me a while to figure out how to build the print menu from the Virtual SD card. I suspect that implementation is still a bit glitchy, depending on what you do to that card directly from [...]

I played with the Creality Sonic Pad a while ago and I think there is also a menu with custom macros in there, that you can enable in the settings. Of course if you let the user trigger any of their custom macros, they should know about the danger of that? Maybe show a warning? I mean we can also trigger these macros from web-frontend with same potential risks.

Thinkersbluff, 2025-05-02:
> I played with the Creality Sonic Pad a while ago and I think there is also a menu with custom macros in there, that you can enable in the settings. Of course if you let the user trigger any of their [...]

I agree that the risk of unexpected outcomes is entirely on the user.  The burden of development, however, falls on me.

If I am going to spend my time developing features for this system, I prefer to integrate useful features with stable compatible code, rather than enabling potentially chaotic outcomes, which would I fear inevitably result in Issues and requests for change.

The Mainsail Console already provides access to user-defined macros. 
If there are missing functions for which local control is desireable, I prefer to deliberately integrate those functions into the design.

Thinkersbluff, 2025-05-02:
> > * Would you also want to be able to set that as your default behaviour (e.g. on SetUp menu)
> 
> I guess that would be kind of surprising if you start a print after a couple of weeks and forget about that feature?

I have a couple of "imaginary user profiles" for whom I try to maintain this interface.

One has a standard set of preferred settings, which he likes to "set and forget", so that he can always trust the system to behave in the same way.  
- This user prefers to form an opinion on whether automatically unretracting the filament after each print is exactly the behaviour they want every time, or a behaviour they do not want.  
-  Upon the introduction of this new feature, they would either leave it Disabled (default for new users) or Enable it, on the SetUp menu.

A Second has a variety of configuration preferences, depending upon the print job.
- This user prefers to "Prepare" their printer for each print job, reviewing the Enable/Disable configurations before selecting "Print"

The Third tends to forget to configure one or more setting before starting their Print and would rather not have to abort the print to go back, fix that setting, and restart the print.
- The Print Tune menu tries to serve this user by providing some control over settings such as this one.

One of the challenges of serving the Third User well is knowing that sometimes, they forget that they have forgotten.
- In the case of the Enable/Disable Runout Sensor function, I put the control on the actual "Printing" page, hoping they notice the current state of the control.  The downside of that approach is that it rapidly consumes the available screen space.

There is not enough screen space left to add a new control for "auto-unload filament at end" on the Printing page
Do you have ideas how that status might be displayed graphically?   

Would it be an acceptable tradeoff to move the LED ON/OFF control to Tune instead, to make room for this one?
Would "Settings" be a better name than "Tune", if some controls on the Tune page are not actually for "tuning the print"?

jensweimann, 2025-05-03:
> Would it be an acceptable tradeoff to move the LED ON/OFF control to Tune instead, to make room for this one?

Sure, I don't use that LED toggle at all personally. But maybe others see it differently. 

> Would "Settings" be a better name than "Tune", if some controls on the Tune page are not actually for "tuning the print"?

How about "More Settings..."? And then there are all the misc. things that aren't that relevant for the print?

Thinkersbluff, 2025-05-03:
I have spawned  sub-issue #58 to discuss and develop the design of a new Enable/Disable control to configure the printer to either automatically unload filament or not, while running the END_PRINT macro, based on the user's setting of one new control.

I have reviewed your posted custom macros and - if you will permit me - I propose to incorporate parts of your custom macros into a new release of the Klipper component of DGUS-Reloaded for CR6, specifically to support this new capability.

NOTE: I may not implement all of your programmed behaviours in the END_PRINT macro as I distribute it, but I do want to ensure that what I implement within the DGUS-Reloaded application will definitely support your needs.

Please review the requirements as I have outlined them in [Issue#58](https://github.com/Thinkersbluff/DGUS-Reloaded_for_CR6-DWIN-SET_Component/issues/58)  and advise whether you anticipate any problems or would like to express preferences for any other implementation details not yet addressed.   It is much faster for me to pivot before I commit design details to code, especially in the DGUSTools display component.

NOTE: "Pass" is ok, too, if you don't want to get this "deeply into the weeds."

Thinkersbluff, 2025-05-26:
The script idea discussed in Issue#63 could be used to identify/extract/build a list of custom macros to be made available to the user.
It should be straightforward to scroll through such a list and "launch" one in a similar fashion to the PRINT function.
Would that work for you?

jensweimann, 2025-05-28:
That would be great. Will there be a special file where we would put our own macros like user-macros.cfg?

Thinkersbluff, 2025-05-29:
I certainly can support a dedicated user_macros.cfg file.  

I can also program an app to search printer.cfg  and all  of the . [included} cfg files in the same config directory for the string "[gcode_macro*" and pull the names of all of those macros into a list. 

Would you want to exclude some macros or to build lists based on specific keys? (e.g. "List all RUN_ABL macros") ?

I do not have access to a Sonic Pad, so I do not know what that interface looks like.
What functionality are you hoping for? (e.g. Sort, Filter, other?)

Thinkersbluff, 2025-05-30:
> maybe cut the macro title if it is too long?

The maximum length of a macro name will certainly be something to manage.   
Klipper will only run a macro if we use its full name, so we would at least have to be able to capture the full name in the firmware.  We may not be able to display the full name to the user, if the name is more than 32 characters long. 

> Klipper web-frontends also show all of them so why not?

I don't like the Sonic Pad approach, where the list is "cluttered" with macros one would never want to launch.  

> User-macro file where the user defines the content

If we require the user to list the names of existing macros to be listed by this function, we could add a section  (e.g.: [custom_macro_list]) to the presets.cfg file, so I know where to find that list and how to parse it. 
I would still have to verify that the listed macros actually existed, but their displayed order could simply match the order in which they appear in the user's list.    

I am not aware of any Klipper feature/standard that would hide macros whose names begin with an underscore character.
Where can I read-up on that?

jensweimann, 2025-05-30:
Afaik the underscore for hiding macros is broadly supported:

Fluidd:
https://docs.fluidd.xyz/customize/hide

Mainsale:
https://docs.mainsail.xyz/overview/features/hide-macros-outputs-or-fans

KlipperScreen:
https://klipperscreen.readthedocs.io/en/latest/Quicktips/#:~:text=without%20much%20explanation-,Hide%20macros%2C%20outputs%20or%20fans,the%20name%20with%20an%20underscore.&text=fan1%20will%20show%20in%20the%20interface%2C%20but%20_fan2%20will%20be%20hidden.

Thinkersbluff, 2025-05-31:
Goes to show, you don't know what you don't know.

Ok, I will add "hide any macros that start with an underline" to the specs for this custom macro function.
Easiest way to achieve that would be to only list macros that the user specifies, e.g. in a custom_macros.cfg file.
I could also put comments in such a file, to give basic info about the existence of the function and any constraints on its use.

When would you want access to these macros, and would it be useful to only present them in the appropriate context?
e.g. Macros to support active printing could be accessed from the Tune menu.  Macros to support SetUp or Prepare workflows could be accessed from their main menus.
NOTE: I don't have the screen real-estate that KlipperScreen or Sonic Pad typically enjoy, so it is harder to imagine always having the Macro menu button visible...

jensweimann, 2025-06-04:
> When would you want access to these macros, and would it be useful to only present them in the appropriate context?

Since we don't know which macros will be in there, it's not possible to know the context. Maybe it would be in the main menu and then in the tune menu?

Thinkersbluff, 2025-06-04:
The core design of the UI is based on grouping controls into the context of Workflows.

I could add a [Home] section, to list macros that should be accessed from a macro button on the Home Screen.

If I had a macro that turned on an enclosure light, then I would list that macro name in every section where I might be performing a particular workflow when I would want to do that. Maybe that particular macro would be needed in every context.

I am not yet sure whether the macro list will be able to scroll, and I don't like the idea of just listing everything, regardless of context.  I have suggested a max of ten macros per context. Hence, I am proposing an approach where the macros that support one of the four workflows can be grouped and listed when needed, like a set of custom menu screens.

If every macro should be listed in every workflow context, the idea of using workflow context to sort and filter is unnecessary complexity.

jensweimann, 2025-06-06:
I like the second image better because there is more space for list items. 

And can we name that section "ABL-Macros" or "ABL" since it's in the Calibrate menu? Will the section names be hardcoded or configurable?

Thinkersbluff, 2025-06-06:
I now have a prototype with which to experiment.

This initial prototype mirrors the current Print screen list of printable files.  You can list as many macros as you like, but the display only shows a maximum of five macro names at a time, one per line.  As you tap the up and down arrows, the list scrolls up and down.  (I have enough screen area to display up to seven names at a time instead of five, it was just faster to set up the initial prototype with five.)

Aesthetically, the current prototype looks strange with lists shorter than five names, because the top item in the list is displayed in the middle of the screen. (It emulates a roller,)
I should be able to reprogram the display to list from the top of the box and to run the highlight up and down the list, but it is much more complex to implement variable properties for each line on the display, rather than hard-coding the display properties and scrolling the data.  If I do change that display design, I will also change the Print page to match it.

The section headings are hard-coded into the routines that locate and build the list of macros to be displayed.
Which section's list to display is coded into the script triggered by the button that switches the display to the MACROS page.

User-specified group headings within each section are not implemented in this version. I can implement those if they are useful, but they do each consume a line that would otherwise display a macro name.  

I need to make a design decision about how to handle a case where the macro name listed in the .cfg file does not match any [gcode_macro <name>] defined in Klipper.  
* I can leave Klipper to flag the problem when Run Macro tries to run it. (Easiest]
* I can filter it out of the list at display time, either silently or with an M118 message to the console, listing the undefined macro names [My preference]
* I can list it but colour that line red and either let Klipper flag it if the user tries to Run it or prevent the run attempt and issue a warning message on the screen. [Complex]

Thoughts?

jensweimann, 2025-06-07:
I would probably just leave the first option. The second option is nice, but could produce questions if the macro is missing and it's not obvious what's wrong. The third option is also nice.

Thinkersbluff, 2025-06-07:
Thanks for the feedback. 
Option 1 is my fallback plan. 
I am experimenting now with the third option, relying heavily on AI to figure out the "how". Whether I succeed or not, it is teaching me some very useful and reusable coding tricks. 
Hoping to "crack this nut"in the next couple of days. 

Would you be interested in "kicking the tires" on what I come up with, before I release it formally? 

If you already have a particular set of custom macros you would like to use, I am willing to incorporate those into my development testing.

jensweimann, 2025-06-07:
I use mostly these:

1. Load- and Unload-Filament. These have my values for how much filament to push/pull and temperature presets
2. Unload-filament-after print enable/disable
3. Turn-off-after-print that will shutdown the system after print, when the nozzle cooled down under 100 °C.  Then my home assistant will turn off the socket automatically. I guess this is highly personal, since it involves home assistant. It uses gcode_shell_command
4. Occasionally light-macro that also uses home assistant with gcode_shell_command

I would gladly kick the tires!

Thinkersbluff, 2025-06-08:
I now have a functional beta version of DWIN_SET 0.4.9 and v1.4.0 of the supporting Klipper components of DGUS-Reloaded, ready to evaluate. 

Since the latest full release is v0.4.7/1.3.8 and this new beta version depends on some elements of the pre-release components 0.4.8/1.3.9,  I will leave this "beta" version on its own dedicated branch, "Custom_Macros_Project", for now.

1. Be sure to backup your system before starting, so that you can restore your system after testing.

2. Overwrite the following files on your system with the 1.4.0 Klipper component files:  t5uid1.py and the contents of the dgus_reloaded folder
[t5uid1.zip](https://github.com/user-attachments/files/20642754/t5uid1.zip)

3. Upload to the Mainsail Machine configuration folder, the latest version of: Related Changes/Creality CR6 Mobo/Custom Klipper host files/For ERA 1.1.0.3 and 4.5.3 MB/[DGUS_Menu_Macros.zip](https://github.com/user-attachments/files/20642735/DGUS_Menu_Macros.zip)

4. Flash your display with DWIN_SET v0.4.9-BETA from this zip file: [DWIN_SET_v0.4.9_BETA.zip](https://github.com/user-attachments/files/20642726/DWIN_SET_v0.4.9_BETA.zip)

You should now be able to edit the new cfg file to tailor the custom macro menus that appear on each screen where the new macros icon appears on the top bar, to the left of the Emergency Shutdown button.  (If I have forgotten some other dependencies and your system does not run correctly after the above changes, I apologize in advance.  Please let me know here and I will try to help you overcome those issues.   NOTE: This version may NOT be "safe" as a "daily driver"... Be sure you can restore your own setup!

Please append your feedback to this issue thread.

Thinkersbluff, 2025-06-09:
Yes, I left behind the button graphic by accident, when I moved the touchpad up to the icon.  I will upload a new DWIN_SET today, fixing that.
The empty list is because the code does not tolerate comments beside the section headings.  If you erase those, or move them above the section headings, onto their own line in the CFG file, that problem goes away.

Thinkersbluff, 2025-06-09:
Here is the updated DWIN_SET, which removes the non-functional macros button from below the four main buttons on the Home Page.

[DWIN_SET_v0.4.9_BETA_090625_1.zip](https://github.com/user-attachments/files/20655157/DWIN_SET_v0.4.9_BETA_090625_1.zip)

Thinkersbluff, 2025-06-09:
Here is the DGUS_Menu_Macros.cfg file that moves the comments and updates the instructions.

[DGUS_Menu_Macros.zip](https://github.com/user-attachments/files/20655279/DGUS_Menu_Macros.zip)

Thinkersbluff, 2025-06-10:
<Internal error during connect: 'T5UID1' object has no attribute 'get_macro_registration_status'

Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted>

Important lesson learned  - FIRMWARE_RESTART does not reload some elements of t5uid1.  Must ReBoot host, to test edits.   

I left a call to that deprecated routine at line 257 of  routines.cfg.  

These edits resolve the error on my system, when I comment-out  these two lines: 

1.  This line (#257) in routines.cfg:   {% do set_variable("macro_registration_status", macro_registration_status) %} 
2.  This line (#299) in t5uid1.py:       'get_macro_registration_status': self.get_macro_registration_status,

Here is an updated t5uid1.zip:

[t5uid1.zip](https://github.com/user-attachments/files/20676715/t5uid1.zip)

jensweimann, 2025-06-17:
I cannot tap on the macro, it doesn't turn green. It beeps but nothing changes. Run Macro also doesn't do anything. It seems that no inputs are recognized, but I hear beep tones from the touchscreen.

jensweimann, 2025-06-20:
Alright, it works now. But I find it pretty unresponsive now. Can we reduce the delay? Where is that constant? I only found CMD_DELAY = 0.02 in t5uid1.py

Thinkersbluff, 2025-06-20:
The switch debouncing delays are not assigned in the python code, they are programmed into a DWIN_SET .bin file.

There is no delay on the macro selection.

The debounced controls already have the shortest programmable delay of 0.1 seconds assigned.

I also find it takes a bit of time to adjust to the debounced UI.  That is one reason I asked for help with testing it.  I may revert it to the previous design.

jensweimann, 2025-06-21:
> Aside from the responsiveness, what is your impression of the functionality of the macro page filtering? Can you access the macros that you want to, from the screens that you want to?

Yes, I think it's great except the unresponsiveness. 

What I noticed: I have one macro defined in lowercase. I also put it inside DGUS_Menu_Macros.cfg as lowercase, and it wasn't recognized. I then changed it to uppercase (only in DGUS_Menu_Macros.cfg), and then it worked. I'm not sure about case sensitivity for macro names, but I would guess they are case-insensitive. Fluidd shows that macro in uppercase, though.

Thinkersbluff, 2025-06-21:
I will run some tests on case, then.  Thanks.

The function compares the macro name in the CFG file to the macro name stored in Klipper.  If the two strings do not match exactly, it says, "No macro selected."  Now that you mention it, it makes sense that since only one comparison is being made, only one version of case is going to match.

I am pretty sure I can convert the case of a string, to fix the compare.  Would you expect that macro name to still be displayed in lower case, in that situation?

jensweimann, 2025-06-21:
I would suggest to display all macros in uppercase since Fluidd also shows it in uppercase. And it also looks nicer.

Thinkersbluff, 2025-06-21:
The quick fix for this issue is to add ".upper()" to line# 1498 in t5uid1.py, as follows:

[image]

Thinkersbluff, 2025-06-21:
To be clear, when you describe the UI as unresponsive, which specific controls are a problem?

jensweimann, 2025-06-24:
Thank you, looks good. 

The unresponsiveness is the time from the tap until the beep/reaction. Sometimes I don't stay long enough on the button for the touch to register and I have to press multiple times.

What I also noticed is that we have now two different kinds of list navigation. We have that list in the print menu where the green/selected item is in the middle and I have to scroll/select with the up/down buttons.

Thinkersbluff, 2025-06-24:
Yes, we have two types of list, for now.
The print list was coded first.  It only has five elements.
With nine macros, I found the Print list approach to scrolling to be too slow.
It will take me a little time to decide how to refactor the Print screen to use the macro list approach.

I understand having to get used to pressing the controls a bit more deliberately with the debounced controls.
 If there are specific controls that you find problematic, please name them.

jensweimann, 2025-06-27:
Well touch inputs are slower now but most slowest is the show macros button and the run macro button is also slow. I mean i'ts not a problem, it's acceptable when you get used to it.

Thinkersbluff, 2025-06-30:
FWIW:  I just uploaded some edited files to the Klipper component repo, in branch "New-At-1.4.1", which refactor the methods that build and display the macro menu pages.  I am hoping they process a tad more quickly for you.  
I also added a short routine to blank the messages line, when you exit from a macros page.

To try the revised firmware, the following files can be downloaded from the repo and used to overwrite the same files from v0.4.0:

t5uid1.py
routines.cfg
vars_in.cfg
vars_out.cfg

jensweimann, 2025-07-01:
Hi, 

thanks, i tried it. Unfortunately it's still like it was before. I have observed the cpu usage with "s-tui" (sudo apt install s-tui) and I don't see any cpu spikes while navigating the touchscreen. The cpu usage is very low 2-7%

I have recorded a video, maybe you could compare that with your performance.

https://drive.google.com/file/d/1wFls9G5XRySJw16YZqyPy8jVWI85vvMa/view?usp=drive_link

Thinkersbluff, 2025-07-01:
Similar to my system.  I was not sure whether the difference in RAM size might have been a factor.

I wonder whether the system is just a bit slow to check for a touch, sometimes.  I know it only refreshes the screen display every 2 seconds.  If it happens to be doing that update when the screen tries to send a "touched" message, there may be a slight delay on the serial interface?

Note: I have extended the touchpad area of the return arrows to about 1" wide, so you don't need to press directly on the arrow icon.

Thinkersbluff, 2025-07-01:
The new touchscreen behaviour has been somewhat of a disappointment, to be sure.  
I am not happy with the DWIN implementation of button activation delays, and the controls I most wanted to debounce (like LED ON/OFF) don't include a delayed action option.

I will reverse that change, and see if things get better.

jensweimann, 2025-07-01:
Nice, works great now! Slight delay on loading the macros page but nothing to complain about. Thank you

Thinkersbluff, 2025-07-02:
I will not rush out a third release today, but FYI I figured out  how to debounce switches in the code, this afternoon.  I have uploaded modified t5uid1.py and vars_in.cfg files to demonstrate how I now debounce the macro screen selection buttons.  
No delay on the first action, but ignores additional presses of the same button for a user-specifiable interval (default = 0.5 secs). 
I also missed a couple of controls when removing delays at DWIN_SET v0.5.1, so I have uploaded a repaired v0.5.2 on its own branch of the repo.