Thinkersbluff/DGUS-Reloaded_for_CR6-DWIN-SET_Component #58: New Feature Request: Add Option to Automatically Unload the Filament when End Print Macro runs

Thinkersbluff, 2025-05-03:
Here is a proposed set of requirements to guide the design/implementation of the proposed option. 
If you have questions or suggestions, please Quote Reply this comment, but delete any of the irrelevant text from the quote.

1. Allow users to Enable or Disable this option:
- As part of printer SetUp 
- As part of the Prepare workflow (Add new screen, "Enable/Disable Options"  below Screen 006)
- From the Print menu (Screen 016)
- Any time during a Print (whether Tune Paused Print (Screen 011) or Tune Active Print (Screen 003) )
- From the Print Finished screen (Screen 004)

2. Make user's choice persistent, as with LED ON/OFF and with Runout Sensor Enable/Disable, using the same variables.cfg file and logic.  
Specifically: Use the variable name "unload_filament_after_print" to store the latest setting.  Use the value "1" to represent ENABLE and 0 to represent DISABLE.

3. Implement precisely the same Enable/Disable control on each display screen where the function is made available 
(i.e. Do not implement this control in different ways on different screens.)

4. Label each instance of this control, "END_PRINT_unload".  Label it with text on the screen, in the same font/colour/size as the "Filament Runout" label on Screen 002.

5. On every page where the Enable/Disable control is implemented, display the currently commanded state of this control as follows*: 
- Illuminate the control (i.e.: fill with Red, Font = white) and display the word "Enabled", when the system is configured to  automatically unload the filament, the next time that the END_PRINT macro runs.
- Darken (i.e. fill with Black, Font = white) and display the word "Disabled", when the system is configured to NOT automatically unload the filament, the next time that the END_PRINT macro runs.
(* CAUTION: The actual state of the control may not match the latest commanded state, if the system is busy processing other commands when this control is activated.  As with LED ON/OFF and RunOut Sensor ENABLE/DISABLE, a user may need to wait as much as a 2-4 seconds before seeing the control confirm their selection.  During such times, the M117 message of requirement 4 should provide users with confirmation that their last selection is being processed.)

6. Test and report the latest commanded state of this control into the Klippy Log (M118) and on the screen (M117) on the following occasions:
- When the Print menu is displayed
- When the Print Finished menu is displayed
- Each time the Enable/Disable control is operated

7. Incorporate a new "If" subroutine into the END_PRINT macro, to test for the currently commanded Enabled/Disabled state of this new control, once the head has been commanded to its "parked" position.
If Enabled, then run the instructions to unload the filament.
If Disabled, then skip the unload and proceed to the next step.

Thinkersbluff, 2025-05-03:
My design goals here are to try to: 

1.  - keep the evolving UI consistent with the current Workflow-Oriented design; SetUp, Calibrate; Prepare; Print.
2.  - maintain the current DGUS-Reloaded application architecture, but also to evolve that architecture if/as necessary to incorporate new features

The first goal guides requirement 1
The second goal guides requirements 2-7

Thinkersbluff, 2025-05-04:
Here is a quick prototype of the change as it could be implemented on Screen 002 - Print Status.
If this is not too "crowded", it would not be necessary (yet) to also implement this control on the "Tune" screens.

[image]

[image]

jensweimann, 2025-05-06:
Looks good to me 👍 Great work man, you rule!

Thinkersbluff, 2025-05-06:
Feature implemented in DGUS-Reloaded for CR6 Display Component v0.4.7 / Klipper Component v1.3.8.

Release still in WIP on 6 May, pending update of the Related Changes files and formal release on the GitHub.
Expecting to complete this release in the next 48hrs.

closed 2025-05-07 by Thinkersbluff