Thinkersbluff/DGUS-Reloaded_for_CR6-DWIN-SET_Component #60: Auto Bed Levelling page showing incorrect heights

Zanadar, 2025-05-11:
Given the following mesh in the SAVE_CONFIG section of printer.cfg...

#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	-0.002500, -0.031250, 0.075000, 0.038750, -0.117500
#*# 	-0.030000, 0.013750, 0.043750, 0.043750, -0.010000
#*# 	-0.106250, -0.056250, 0.018750, 0.036250, -0.066250
#*# 	-0.106250, -0.011250, 0.018750, 0.058750, -0.068750
#*# 	-0.085000, 0.006250, 0.018750, 0.076250, -0.071250

The default profile on the ABL page is showing down the left column (starting with the top) (for left hand side of bed):
-0.085
0.000
-0.106
-0.030
-0.002

0.000 should show -0.106... all other heights are displaying correctly (with rounding).

Zanadar, 2025-05-11:
On further review... this position (2nd height from top in left hand column) is always showing 0.000 for every mesh.

Thinkersbluff, 2025-05-21:
Good eye!
That particular position was incorrectly addressed.

Corrected at v0.4.8.
Thank You!